Mellow-3D/Fly-SB2040 #4: Wrong temperature values returned for onboard NTC100K sensor

lemmingDev, 2022-12-26:
Hi

When using the onboard NTC100K to check the temp with:

```
[temperature_sensor Warehouse]
sensor_type = ATC Semitec 104GT-2
sensor_pin = sb2040:gpio26
```
the values is too high.

When I set 
```pullup_resistor: 10000```
then the value is better

Did you guys put a 10k resistor instead of 4.7k?

Thanks

djessup, 2022-12-26:
I have the same issue. A board I bought a month or two earlier worked as expected with a default (omitted) `pullup_resistor` value.

arielbernal, 2023-01-09:
Same issue here.... the NTC100K values are way too high, similar to the mcu temperature. 
I would be nice if Mellow could answer the question

pbutler, 2023-01-18:
I am having the same issue... but even 10k pullup doesn't seem to be 100% right for me.  Room is ~16C but the thermistor in the unheated chamber reads as 22C

lemmingDev, 2023-01-18:
I gave up and put a standard hotend thermistor in the z drag chain connected to the Octopus board directly to use for the enclosure temperature sensor

arielbernal, 2023-01-18:
> I gave up and put a standard hotend thermistor in the z drag chain connected to the Octopus board directly to use for the enclosure temperature sensor

I gave up as well, I used the z drag chain as too

djessup, 2023-01-18:
@mellow-3d Could you please give some input on what is going on with your SB2040 thermistors here?

Villiem, 2023-01-25:
Same problem

VooDooSi, 2023-01-27:
I have the same problem. I dont really care for onboard thermistor, problem is i also get negative temperature for extruder. So far i flashed the board, connected extruder thermistor, fans and leds. Even with everything disconnected all the temperature readings are negative.

pbutler, 2023-01-27:
@VooDooSi:  I came across this document someone posted either to the Voron forums or reddit that mentions negative temps: https://docs.google.com/document/d/1vqLZFZ6slT3QIw_dWkraFHuKx9kYC-jz_46KikXdVu4/edit

VooDooSi, 2023-01-27:
@pbutler, thank you. I saw this already yesterday when i flashed the board without connecting anything else and i got negative temperature for pcb ntc. I didnt even come to connecting endstops :)

muditguptacode, 2023-02-07:
I initially thought I was good after switching the pullup value to be 10000 but no, the values "look better" but still don't seem to reflect the actual temperature. 
@Mellow3D do you actually intend to address this in any way or provide a potential fix? I'd take a scope to the board myself and diagnose but I really don't want to disassemble my printer at the moment.

thiagolocatelli, 2023-02-14:
@jaysuk Do you mind helping the community and answering to this issue?

jaysuk, 2023-02-14:
i would if i knew the answer

thiagolocatelli, 2023-02-14:
@jaysuk I saw that you are the one made all the commits to this repo, so I apologize if you are not the correct person. If you work with Mellow, can you please reach out to Mello and have someone help with this issue?

thiagolocatelli, 2023-02-14:
@jaysuk Thank you so much

Mellow3D, 2023-02-14:
Yo please stop sending me emails, clearly i am not supposed to be in this
loop 😃

Peace

rbrtwtrs, 2023-04-04:
I am also seeing high temperatures but they seem legit. The TPS5430 Buck Converter or the adjacent TF7409 FET is getting extremely hot. Like hot enough to burn your finger hot. I see that rev 2 of the board addresses a potential short but I cannot find information about where the short happens. The board still works but will lose comms once it gets above 70C.  I have ordered a rev2 board and hope that fixes the issue. I am using the Rainbow Barf LEDs and the power draw is considerably worse when they are connected. 
I suspect the power dissipation is lacking and it may require an external 5V supply to prevent overheating. If Rev2 board has the same problem I will explore that more.

rbrtwtrs, 2023-04-04:
From what I see the 24V supply current to the Mellow fly runs at 35ma or nearly 1W just idling. With that much power, the board will heat up a fair bit. If I connect the Neo Pixel string then the power goes up by 10x and it is going to get extremely hot. Being in a 60+ degree enclosure and running the stepper driver it is going to toast itself in no time. I suspect this is not going to work without supplying the 5V externally.

arielbernal, 2023-04-05:
@rbrtwtrs, I'm running the 2040 with a small fan, and the temperature very stable a little bit over 60C even tho the enclosure temperature is ~60C with the plate at 105C. Before I added the fan the board shut down at ~80C, at about 5-10min into the print.
I have similar issues with the NeoPixel leds, even if everything is cold. I thought that the compute power of the r2040 was not enough to handle the fancy effects, but it could be that the leds are under powered, because it happens when heating or printing.

rbrtwtrs, 2023-04-05:
I did not investigate the causes of the relatively high idle current. The 2040 should only be drawing 40mA off the 3.3V Supply when running full tilt. The Stepper Driver is not a low power device but it all seems very power hungry. I see that the V2 board has a fan. I have put a heatsink on both the 2040 and Stepper Driver but that device that is the hottest is the Buck Converter IC and there is no clearance to put a heatsink on it. It would make sense the replace the plastic standoff spacer under the Fly with a metal piece that could act as a heatsink for the Buck Converter. 
Precision is not one of the qualities of the RP2040 and it states in the datasheet that the internal temperature sensor requires user calibration. Apparently, the voltage reference is pretty low quality and drifts.
I have a non-standard Neo Pixel string as the two nozzle NeoPixs are high-brightness all-white versions to illuminate the nozzle for my nozzle camera. I will have to run an external 5V wire or a separate 5V regulator in order to run those.

rbrtwtrs, 2023-04-05:
@arielbernal Did you modify the Stealthburner shroud to mount the fan? Do you have an stl for that?

rbrtwtrs, 2023-04-05:
I did not look into the possibility that the RP2040 may be being overclocked and if so it may be using far more current than the datasheet says.

arielbernal, 2023-04-06:
@rbrtwtrs, Yes I use a mod for the Stealthburner but don't recall where I got it. I'll push it to my github. The cover is also big enough to allow for a heatsink over the 2040 and the TMC 2209. 
You mention that you run the LEDs the power goes up and the board heats up. I wonder why as the 5V are coming from your PSU and the ASM11117-3v3 shouldn't heat up much for the pi2040 and comms. 
One of the reasons for the NTC100K for bad readings is because it's on the board. I'm giving up on the NTC100K and running a separate one on the Z drag chain

arielbernal, 2023-04-06:
@rbrtwtrs Here it is https://www.printables.com/model/300272-sb2040-fan-cover

rbrtwtrs, 2023-04-06:
The SB2040 only gets 24V and has a buck converter that supplies the 5V, there is an additional 3.3V regulator running from the 5V and that is what the 2040 runs on. I have added an additional 5V regulator https://www.amazon.com/dp/B0B779ZYN1?ref=ppx_yo2ov_dt_b_product_details&th=1
It 
[image]
It mounts on top of the Extruder Motor Connector and the power returns line up great with the flanges on the Extruder Connector so it is quite secure. I run the 5V output wire directly to the 5V leg of the 8 pin cover interface and have isolated it from the 2040 boards 5V. Interestingly the onboard 5V Regulator still runs abnormally hot.
I have just received a V2 SB2040 kit but there is no documentation or version-specific firmware available yet. Parts have moved around between the versions and most importantly the board has a completely different 5V regulator design. will be interesting to see if there are improvements to the heat issues. I suspect there is a significant problem with the V1 design.

rbrtwtrs, 2023-04-06:
I decided to completely bypass the TPS5430 5V Buck Converter and run the board from the 5V regulator I added. This completely solves the high-temperature issues. The onboard thermistor temp is now reading close to ambient and the RP2040 internal temp stays under 40C. From looking around at other RP2040 projects that appears to be a reasonable value. The external regulator does not get hot at all when my NeoPix string is not connected and barely warms up when it is connected.
I just lifted pin 8 from the TPS5430 IC to disable it. Would probably be best to remove the IC completely along with the inductor L1. I cannot find any outright errors in the design other than perhaps the inductor should be a 15uH rather than the 10uH one they use. I don't really have time or desire to look at it in more detail. The converter is rated to 3A so should not be getting warm at all. There must have been a reason that Mellow abandoned that design in V2.

MyStoopidStuff, 2023-04-21:
@rbrtwtrs I'm putting together my SB with an SB-2040 V1 board, and wanted to say thanks for the insights on the buck converter.  When you said the SB-2040v1 was drawing nearly a watt at idle I was a bit shocked, since I read a Pi-2040 will only consume around 25mA (which is nothing in comparison). 

I looked up the datasheet for the TI TPS5430DDAR buck converter,  and used the TI webench tool, which is linked from the [datasheet](https://www.ti.com/lit/ds/slvs632j/slvs632j.pdf?ts=1682047075763) on p1 bottom of section 1.  When I ran the webench tool for the part (using 22-26v for the input range, which was just a guess), I noted several differences between what was chosen for the SB-2040 Version 1, and what the webench design showed.  I also noticed that they used a 1N4007 diode instead of a Schottky diode that is specified, and like you noted, the inductor is 10uH and not 15uH.  The output cap was 220uF (6.3V) in the SB-2040 v1 schematic, vs 330uF (6.3V) from the webench example I tried.  The input cap appeared the same (two 4.7uF capacitors).  

I'm no expert on any of this stuff though, so it would be interesting to know from somebody who is knowledgeable of buck converter designs, if the difference in components (vs a design generated by TI's webtool), is possibly part of the trouble.  I haven't even programmed this board yet, and I'm having second thoughts about it lol.

rbrtwtrs, 2023-04-21:
I noticed the same discrepancies with the component values. The fact that they switched to a completely different regulator on V2 makes me think there was more to it. Switching Power Supply design is a much trickier thing to get right than many people realize. The reactance of the loads can completely throw things off. Perhaps the stepper motor windings? Dunno but it should be able to handle it easily and it obviously doesn't. They are trying to keep the parts cost down and every penny counts so using a 10uH inductor and cheap generic diode helps. I think using a 6.3V Cap is really pushing your luck on a 5V switcher.
I haven't gotten around to using my V2 board yet and am just running with the add-on 5V regulator board. I saw someone else was doing the same thing but just running their Neopixels on a secondary supply. From what I see the onboard Buck still runs red hot even with just the 2040 loading it.

MyStoopidStuff, 2023-04-21:
@rbrtwtrs Haha yeah I forgot about the rule of thumb relating to capacitor sizing, I guess they did too (though I would bet the footprint was part of the reason).  I just spent some time looking for possible replacement SMD parts for a diode and inductor to have on hand to swap, but soon realized that finding them in the same or similar footprint was not gonna be easy.  

It does look like a different buck design on the V2.  I just looked at the V2 schematic which I somehow could not view earlier and it's a different part.  The V2 buck does away with the diode and uses a smaller inductor.  I hope the V2 works out for you. 

[image]

JanB97, 2023-05-04:
My V2 is just as wrong as the V1 I had before it blew up.
I wonder if it just actually heads up because the board gets warm from the rp2040 and the tmc.

MyStoopidStuff, 2023-05-04:
@JanB97 That is not good news, but thanks for sharing.  I checked with one seller of the V2's and they said they had no complaints, but every data point helps.  You may be able to isolate the source of the heat, since the 5V buck is on the back of the board near the inductor, while the stepper driver and RP2040 is on the right side front.  If you have one, a non-contact thermometer may work to find out if the heat is from the buck converter or the other chips.  On my V1 they sent one heatsink, but I got another 8.8x8.8x5mm heatsink so I could cool both the RP2040 and the stepper driver.  There is also a fan cover which looks like it could help with temps, and has the added benefit of allowing better access (https://www.printables.com/model/438309-mellow-sb2040-stealthburner-cover-with-fan).  I dropped my V1 and bought a BTT SB2209 and am working on a remix of that fan cover design for the BTT now since it allows much better access to the thermistor than stock.

JanB97, 2023-05-04:
I have this one on mine: https://www.printables.com/de/model/353867-voron-stealthburner-sb2040-cover
I now just upped the pullup resistor a bit to 5400 so it actually matches pretty well with another thermistor I put in while heated.
It's of by around 8 degrees when not heated, but I am more concerned with somewhat accurate readings while the printer is in use.
I also have the included heatsink on the tmc and a little raspberry heatsink on the rp2040.
I was thinking about going with the BTT after the v1 died for no reason, but I didn't feel like working on the wires again, that's why I have gone with can in the first places.

MyStoopidStuff, 2023-05-05:
@JanB97 Interesting, did you replace R16 to alter the thermistor values (I thought the 4.7k resistor there was a correct value and appears the same between V1 and V2)?  I just wired up my BTT SB2209 yesterday, and was surprised that I could crimp the tiny connectors using an iCrimp IWS-3220M, and the ones that I could not, I used some pre-crimped wires which I soldered in, but yeah the tiny pins are no fun, no perfect solution for a simple CANbus solution yet it seems.

JanB97, 2023-05-07:
No I didn't change any resistor.
I believe 4.7k is the correct value, I just wanted it to be more accurate and compensate for the heat thats there from the rest of the board I guess.

djessup, 2023-05-07:
I'm starting to wonder if the values could be correct all along and it's reading high because of how much heat is soaked into the board from the 5V regulator or whatever the consensus is re generating heat?

rbrtwtrs, 2023-05-07:
A good way to see where the source of heat is (short of using a thermal
imager) is to use a can of Freeze Mist to frost the board. The frost will
rapidly go away at the exact source of heat. You can also hold a can of Air
Duster upside down and get a freeze mist from it.

JanB97, 2023-05-08:
Well just seeing the temp of that rp2040 even with a heatsink and probably also the heat from the TMC should be enough to heat up that board. It's not really big enough to space out the thermistor from that.
I am pretty sure it's not reading wrong, it's actually just heat from other components.

rbrtwtrs, 2023-05-08:
Heatsinks and modded shroud w fan are pretty much a must do if you are
running a heated chamber. Rev 1 boards regulator gets hot enough to ignite.
I have not tried rev 2 yet.

EmJay276, 2023-08-11:
My V1 blew up today too, the capacitor and the 5V regulator seem to have failed.
I also noticed my neopixels did not work the last 3 days before the failure, but my tap at the 5V worked fine. Perhaps the voltage has dropped? The tap only needs 3.3V to my knowledge.