Mellow-3D/Fly-SB2040 #5: inductive probe interfere extruder temperature reading on SB2040

Dual-Energy, 2023-01-03:
Hello, I have problems with my inductive probe (Pinda, 5V) on a SB2040. 
As soon as the probe supplies a signal, the temperature signal of the extruder (PT1000) is disturbed. The temperature seems to jump +/- 30°. Especially on the first layer the probe is always triggerd an brings Klipper to aboard the print du to "extruder does not heat on expected rate". 
Tried the available GPIO`s  from the other endstops but without success. 

Does anyone else have this problem? Is there a workaround? 

Thank you

jaysuk, 2023-01-03:
sounds like its more interference than anything.
do your PT1000 cables run next to the pinda cables?

Thanks,

Jay

Dual-Energy, 2023-01-03:
No both cables  (PT1000 and Pinda) are not even close together. Tried to use an extra capacitor (10NF and 100NF) parallel to the PT1000 pins but without any success. Next is to check wether it works with external 5v.

cscashby, 2023-05-14:
I have the same issue - @Dual-Energy let us know here if you manage to fix the issue. For now I'm going to go with Klicky.

cscashby, 2023-05-14:
OK, solved it. Putting the probe in the 5v port (gpio28) will fail. Using the HV port (gpio25) it works fine. @Dual-Energy this may have been your problem? The issue is in my board at least it's on the front (right-angle connector). The picture in the docs shows it as a straight connector.