Mellow-3D/Fly-SB2040 #6: How to connect BL-Touch via SB2040 ?

Gobol, 2023-01-06:
Hello!

Trying to configure Fly_SB2040 with BlTouch, but no luck yet.

What's the best idea to connect with BlTouch ? 

So far I've got: 
```
blt_signal (yellow) on gpio16 (PWM0)
blt_probe (white) on gpio29 (ENDSTOP2)
```

accordingly .cnf setup as follows: 

```
[bltouch]
##	Bltouch - If you use this section , please comment the [probe] section
##	More infomation at : https://www.klipper3d.org/BLTouch.html
##	This bltouch is not used for Z height, only Quad Gantry Leveling
##	FLYSB2040 - ENDSTOP 2pin
sensor_pin: sb2040:gpio29
##	FLYSB2040 - PWM0
control_pin: sb2040:gpio16
x_offset: 0
y_offset: 25.0
#z_offset: 0
speed: 5
samples: 3
samples_result: median
sample_retract_dist: 2.0
samples_tolerance: 0.01
samples_tolerance_retries: 2
probe_with_touch_mode: True
stow_on_each_sample: False
```

jaysuk, 2023-01-06:
you may be able to use the PWM output but I think ultimately the SB2040 just isn't designed to drive a bltouch because really, who would fit one to a voron?

Gobol, 2023-01-06:
Why wouldn't one ? 

Anyway, I got it to work using PWM as signal and ENDSTOP1 as pull-up'd probe.

closed 2023-04-15 by jaysuk

emessage, 2024-02-26:
This is a very old topic but I was wondering if you could post your final 

> Why wouldn't one ?
> 
> Anyway, I got it to work using PWM as signal and ENDSTOP1 as pull-up'd probe.

I know this is a very old post but could you post your final config?  I tried to use gpio16 for signal and 29 for probe but can't get it to work.