Mellow-3D/Fly-SB2040 #7: Documentation canboot firmware flash is wrong

treeific, 2023-02-13:
The documentation here
https://mellow.klipper.cn/#/board/fly_sb2040/flash?id=_2-%e7%83%a7%e5%bd%95%e5%9b%ba%e4%bb%b6
In English
`burning

cd ~/klipper/
make flash FLASH_DEVICE=2e8a:0003
Copy to clipboarderrorCopied
Executing the above command may`

Should instead be
`burning

cd ~/canboot/
make flash FLASH_DEVICE=2e8a:0003
Copy to clipboarderrorCopied
Executing the above command may`

kluoyun, 2023-02-13:
I don't quite understand what you mean, the document here is to burn klipper firmware, not Canboot. Why enter the ~/canboot directory.

treeific, 2023-02-13:
I should have included more details.
I see, you're right, that is for klipper firmware. BUT, this canboot document (link) references that doc:
https://mellow.klipper.cn/#/advanced/canboot

`[2. Burn CanBoot boot firmware](https://mellow.klipper.cn/#/advanced/canboot?id=_2-%e7%83%a7%e5%bd%95canboot%e5%bc%95%e5%af%bc%e5%9b%ba%e4%bb%b6)
1. SB2040

SB2040 burning is the easiest, it is the same as normal burning firmware, but the burning is not the klipper.uf2 file, but the canboot.uf2 file

Please refer to the burning steps: [SB2040 firmware burning](https://mellow.klipper.cn/#/board/fly_sb2040/flash?id=_2-%e7%83%a7%e5%bd%95%e5%9b%ba%e4%bb%b6)`

I was able to figure it out by assuming "but the burning is not the klipper.uf2 file, but the canboot.uf2 file" meant to cd ~/canboot

Please let me know if I need to explain more.

Thanks!

closed 2023-04-15 by jaysuk