Mellow-3D/Fly-SB2040 #9: BL-Touch requires removal of series resistance

hvraven, 2023-05-21:
After some debugging I finally managed to get a BL-Touch (genuine V3.1) working on a Fly SB2040 V1. It is not possible as documented and it would be nice if you could update the documentation.

The BL-Touch is connected as recommended (using `ENDSTOP_0` for the servo and `ENDSTOP_1` for the sensor). To get it operating I had to remove the 1k series resistance at `ENDSTOP_0` (R33) and replace it with 0 Ohms. Otherwise I only got about 2.3V on the GPIO which was not enough for BL-Touch. I also more or less accidentally removed the pull-up (R31) while soldering, which is not necessary for this application.

dward, 2023-05-26:
@hvraven I had a similar issue. I worked around it by running a lead to one of the unused PWM fan pins. I'll try replacing the resistor as that's a cleaner solution.

jeor112, 2023-06-04:
Hi
I'm struggling with the same problem that BLTouch doesn't work with the SB2040 V1.
I have mounted BLTouch as described in their documentation (LIMIT_1 and LIMIT_2 - GPIO28 and GPIO29).

I use HV-IN (LIMIT_0, GPIO25) for the X-endstop switch, where I have set pullup (^) in Klipper's printer.cfg.

Where can I find R33? I assume it's for LIMIT_2 (GPIO29)? 
Couldn't find the print layout with the component location.

Really appreciate any help!
Thanks in advance