Mellow-3D/Fly-SB2040 #15: Thermistor Readings

jamrassic, 2024-09-15:
Hello,

I have two SB2040 boards that have started to have issues with their thermistor readings. One is a v1 and the other is a v2.
I'm installing a new Revo heater and everything is working fine until I move the thermistor wires and then the reading drops down, for both the thermistor port and the built in thermistor on the board. Afterwards, with the thermistor disconnected, the unit reads out -93.8C on the external thermistor, and ~-19C on the built in thermistor. 

Also, when running "~/katapult/scripts/flash_can.py -i can0 -q" the board comes up as a Katapult node prior to the thermistor failure but does not show up as a node after the failure. Klipper never loses communication with the board.

Any help would be great. The v2 board was my backup and I need to get this printer back up and running.

Thanks

jamrassic, 2024-09-17:
Hello,
Just wanted to add to this issue that I was able to cannibalize another printer I have for the toolhead board and that one also failed with a brand new thermistor. It seems like the TMC driver is far to close to the exposed leads of the CAN connector and the provided adhesive for a heat sink let go and might have touched the exposed leads of the CAN header. This for some reason has destroyed the internal reference used by the board for thermistor readings. I have found no other failure of the board aside from the temperature readings. 

The boards do not have part designators on their silk screens so trying to trouble shoot this issue is very frustrating. Are there PCB files somewhere other than this repo?

Thank you,

kullwarrior, 2024-09-20:
I'm also having similiar issue, with thermistor resistance set to 1k and pt1000. Using semitec 104gt2 and 4.7k seemed to be the solution.

techmaster242, 2026-01-11:
I'm having all kinds of issues on 2 printers using the SB2040v2. They've been rock solid printers for 2+ years. I just upgraded my Phaetus Rapido 2 hotends to the 2F model which uses a PT1000 thermistor. I added the jumper to the board on both printers and set the pullup resistor setting in klipper to 1000 ohms. On one printer it panics because it thinks the hotend went above 270 as soon as the nozzle touches the hot bed. On the other printer it prints fine for a while, but after it's a few millimeters off the bed it stops extruding. I just came across this and it sounds like the same thing I'm experiencing. Has anybody found a fix for this yet? Does v3 fix this issue? Or does it come down to use a different hotend or try a different brand of board?