Mellow-3D/Fly-SB2040 #17: Correct accelerometer config for 2040 V3 Pro

runtpacket, 2025-01-26:
Using the config from docs site as follows;
#####################################################################
#      Accelerometer
#####################################################################
[lis2dw]
cs_pin: SB2040:gpio12
spi_software_sclk_pin: SB2040:gpio2
spi_software_mosi_pin: SB2040:gpio3
spi_software_miso_pin: SB2040:gpio4

#--------------------------------------------------------------------
[resonance_tester]
accel_chip: lis2dw
probe_points: 175, 175, 20
min_freq: 5
max_freq: 133
accel_per_hz: 75
hz_per_sec: 1

executing ACCELEROMETER_QUERY returns

Invalid lis2dw id (got ff vs 44).
This is generally indicative of connection problems
(e.g. faulty wiring) or a faulty lis2dw chip.

c1000mg, 2025-02-21:
I'm using the same configuration, but i noticed there is not axes_map: x,z,y, if you notice the direction if the arrow in the SB2040 v3 doesn't match the axes_map

runtpacket, 2025-05-28:
Thank you. I added to my notes to test, right now I'm back to an SB2040V2